hku-mars/FAST-LIVO2 #290: camera标定参数有疑问

asurasoft, 2025-05-22:
我使用的MV-CA013-A0UC和MVL-HF0628M-6MPE，标定出来的K和P值都是600多，300多，摄像机的像素大小设置是1280x1024，使用的7x4的标定板，8cm，这是为什么，是相机的参数设置有问题吗？

mono pinhole calibration...
*** Added sample 85, p_x = 0.538, p_y = 0.289, p_size = 0.271, skew = 0.167
D = [-0.1289158097430596, 0.20771136670738508, -0.005168046403593508, 0.005027243282405017, 0.0]
K = [661.1176185910632, 0.0, 325.4537100774287, 0.0, 659.468568340089, 228.94939774725097, 0.0, 0.0, 1.0]
R = [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P = [649.4614868164062, 0.0, 327.65628256825585, 0.0, 0.0, 648.3731079101562, 226.45834851089603, 0.0, 0.0, 0.0, 1.0, 0.0]
None
# oST version 5.0 parameters

[image]

width
640

height
512

[narrow_stereo]

camera matrix
661.117619 0.000000 325.453710
0.000000 659.468568 228.949398
0.000000 0.000000 1.000000

distortion
-0.128916 0.207711 -0.005168 0.005027 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
649.461487 0.000000 327.656283 0.000000
0.000000 648.373108 226.458349 0.000000
0.000000 0.000000 1.000000 0.000000
:

asurasoft, 2025-05-22:
> 你是用的除以2的分辨率标定的吧： [image]
> 
> width 640
> 
> height 512

可能是的，但是我执行标定程序的时候，是这样
`rosrun camera_calibration cameracalibrator.py --size 7x4 --square 0.08 image:=/left_camera/image`
并没有地方指定分辨率的缩放，是摄像头启动的时候，参数指定缩放的吗？

asurasoft, 2025-05-22:
另外这个标定结果和camera_pinhole.yaml中如何匹配

```
cam_model: Pinhole
cam_width: 640
cam_height: 512
cam_fx: 431.795259219
cam_fy: 431.550090267
cam_cx: 310.833037316
cam_cy: 266.985989326
cam_d0: -0.0944205499243979
cam_d1: 0.0946727677776504
cam_d2: -0.00807970960613932
cam_d3: 8.07461209775283e-05
```

xuankuzcr, 2025-05-23:
K = \begin{bmatrix} cam\_fx & 0 & cam\_cx \\ 0 & cam\_fy & cam\_cy \\ 0 & 0 & 1 \end{bmatrix}
\text{distCoeffs} = [cam\_d0, cam\_d1, cam\_d2, cam\_d3]

asurasoft, 2025-05-23:
> K = \begin{bmatrix} cam_fx & 0 & cam_cx \ 0 & cam_fy & cam_cy \ 0 & 0 & 1 \end{bmatrix} \text{distCoeffs} = [cam_d0, cam_d1, cam_d2, cam_d3]

谢谢