hku-mars/FAST-LIVO2 #291: hesaipandarXT32激光雷达

FairyXrrrr, 2025-05-23:
适配禾赛PandarXT32这款激光雷达需要修改代码吗？外置6轴imu的噪声需要考虑添加吗？

xuankuzcr, 2025-05-23:
不用改代码，但LiDAR的噪声参数要调整。我上传下Hilti数据集的参数，他们用的是Hesai XT32。

FairyXrrrr, 2025-05-23:
好的，谢谢郑博

chongyu2021, 2025-06-18:
> > Hilti没提供真值，但是可以把结果上传到[官网](https://submit.hilti-challenge.com/)，官方会提供评估结果
> 
> 感谢郑博的回复，我看到分数以及几个指标的图，但是rmse只能根据图大致估计大小，没有具体的rmse结果，这个是正常的吗？还是我没有找到可以查看具体值的地方
> 
> [image]

同问

zhangqi-sudo, 2025-07-30:
> 不用改代码，但LiDAR的噪声参数要调整。我上传下Hilti数据集的参数，他们用的就是Hesai XT32。

  nh.param<double>("lio/beam_err", voxel_config.beam_err_, 0.02);
  nh.param<double>("lio/dept_err", voxel_config.dept_err_, 0.05);
请问是改这两个参数吗，应该改为多少呀，参考什么呢，谢谢！

zhangqi-sudo, 2025-08-01:
Preprocess::Preprocess() : feature_enabled(0), lidar_type(AVIA), blind(0.01), point_filter_num(1)
{
  inf_bound = 10;
  N_SCANS = 6;
  group_size = 8;
  disA = 0.01;
  disA = 0.1; // B?
  p2l_ratio = 225;
  limit_maxmid = 6.25;
  limit_midmin = 6.25;
  limit_maxmin = 3.24;
  jump_up_limit = 170.0;
  jump_down_limit = 8.0;
  cos160 = 160.0;
  edgea = 2;
  edgeb = 0.1;
  smallp_intersect = 172.5;
  smallp_ratio = 1.2;
  given_offset_time = false;

  jump_up_limit = cos(jump_up_limit / 180 * M_PI);
  jump_down_limit = cos(jump_down_limit / 180 * M_PI);
  cos160 = cos(cos160 / 180 * M_PI);
  smallp_intersect = cos(smallp_intersect / 180 * M_PI);
}
如果用禾赛XT32，这些参数除了N_SCANS 是32线，其他需要改动吗，请问有人知道吗？