hku-mars/FAST-LIVO2 #293: mid360+广角鱼眼相机建图上色精度

ddenglina, 2025-05-23:
> 是的，一般希望LiDAR和相机的FoV重合得越多越好，这样其实标定也比较容易。Mid360更适合搭配广角鱼眼相机，而Avia则更适合搭配针孔相机。选型没问题的话，两种LiDAR都能得到不错的效果。 

 _Originally posted by @xuankuzcr in [#50](https://github.com/hku-mars/FAST-LIVO2/issues/50#issuecomment-2621556802)_

郑博好，我尝试使用mid360+广角的鱼眼相机，运行fast-livo2的效果上色精度有较大偏差。

已经修改camera.yaml 中的cam_model: PolynomialCamera。

粉色是地面的颜色，墙面是白色的。从建图效果上来看墙面的一部分也上成粉色了。

是因为鱼眼自身的畸变原因吗？

[image]

ddenglina, 2025-06-17:
> 不是，感觉内外参都没有标好

谢谢郑博回复，确实是内参和外参没标好，mid360+广角鱼眼相机建图上色精度不准的问题已经解决了。

dandelion-chu, 2025-09-17:
> _Originally posted by [@xuankuzcr](https://github.com/xuankuzcr) in [#50](https://github.com/hku-mars/FAST-LIVO2/issues/50#issuecomment-2621556802)_ [...]

您好，请问您是用什么方法标定鱼眼相机内参的呢？能指教一下么？谢谢，我这边也是一直上色感觉有问题~~