hku-mars/FAST-LIVO2 #295: 关于时间戳对齐，求解

Dora-ye, 2025-05-25:
我de 设备是mid360和海康A0UC相机，硬同步，已经做过了相机标定和相机雷达联合标定，可以稳定启动。

但是，我录制了很多了rosbag包，imu，lidar，image很难做到像您公开数据集那样漂亮的对齐，总是有零点零几秒的偏移，我通过修改img_offset和imu_offset的值，观测时间戳也是有零点零几偏差秒😭，请问老师，这个影响大吗，长时间使用会不会飘的很严重，或者我该怎么改善，谢谢

[image]

[image]

iorileslie, 2025-07-15:
> 请问您现在是解决了这个问题吗?我也遇到了相似的情况

可以看一下https://github.com/cchester25/FAST_LIVO2_Noted的论文解析-_扫描重合并_中有提到，使用郑博的硬件同步方案需要设置img_time_offset: 0.1

Li-shuangyi, 2025-07-15:
> > 请问您现在是解决了这个问题吗?我也遇到了相似的情况
> 
> 可以看一下https://github.com/cchester25/FAST_LIVO2_Noted 的论文解析-_扫描重合并_中有提到，使用郑博的硬件同步方案需要设置img_time_offset: 0.1

我设置了这个参数之后,图像和lidar间仍然有30ms左右的偏差,请问这是正常的吗