hku-mars/FAST-LIVO2 #298: 在启动相机mvs_ros_driver驱动时报错MV_CC_OpenDevice fail! nRet [80000203]

luliang520, 2025-05-26:
您好，我用CU013-A0UC相机在打开客户端的同时，启动mvs_camera_trigger.launch文件报错如下，请问一下是什么原因？SerialNumber调了，然后PixelFormat几种都试了但还是报错。

[device 0]:
Device Model Name: MV-CU013-A0UC
SerialNumber: 
MV_CC_OpenDevice fail! nRet [80000203]
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of: /opt/ros/noetic/lib//libcompressed_depth_image_transport.so
[mvs_camera_trigger-2] process has died [pid 8989, exit code -6, cmd /home/nvidia/demo/mvs_ros_driver/devel/lib/mvs_ros_driver/grabImgWithTrigger /home/nvidia/demo/mvs_ros_driver/src/mvs_ros_driver/config/left_camera_trigger.yaml __name:=mvs_camera_trigger __log:=/home/nvidia/.ros/log/0c33b43e-3a10-11f0-b259-3c6d66186c78/mvs_camera_trigger-2.log].
log file: /home/nvidia/.ros/log/0c33b43e-3a10-11f0-b259-3c6d66186c78/mvs_camera_trigger-2*.log
[mvs_camera_trigger-2] restarting process
process[mvs_camera_trigger-2]: started with pid [9025]

NutrientFastLine, 2025-05-28:
海康mvs libusb库的问题，参考[https://github.com/xuankuzcr/LIV_handhold/issues/56#issuecomment-2804167084](url) #

NutrientFastLine, 2025-05-28:
海康mvs libusb库的问题，参考解决方法https://github.com/xuankuzcr/LIV_handhold/issues/56#issuecomment-2804167084