hku-mars/FAST-LIVO2 #302: 移动设备时重建效果差

jiatianluo, 2025-05-30:
我在扫描环境时，当设备静止时，可以取得不错的效果，但是移动设备后，尤其是移动路线出现重复后，图就会变得很模糊，请问是为什么，是内外参标定的问题吗

jiatianluo, 2025-05-30:
> 看描述很有可能是同步没有做好，而不是内外参的问题

我硬件同步和软件同步都有，硬件通过stm32进行同步，软件层面主要是通过时间戳进行同步，您是用什么方式实现同步的

thetengda, 2025-05-30:
你是用的是郑博LIV_handhold那套硬件吗，用avia要设置img_time_offset: 0.1，这个问题我之前也遇到了

jiatianluo, 2025-05-30:
我使用的是livox mid70雷达，海康的相机，另外单独买了一个imu，img_time_offset需要怎么设置啊

JoeyYilinWang, 2025-06-05:
我是直接连接笔记本跑的，不是用rosbag。看时间戳是完全同步的，所以您的意思是即使时间戳同步，也需要设置img_time_offset: 0.1？

linyuxiu130, 2025-06-07:
您好我也使用的mid70，也是碰到了同样的问题，如果有办法解决可以方便告诉我吗，谢谢

cwh01, 2025-06-07:
> 您好我也使用的mid70，也是碰到了同样的问题，如果有办法解决可以方便告诉我吗，谢谢

Hello, I am using mid70 to reproduce the error when using the livox_ros_driver2 driver. The radar has not been published. Have you encountered this situation before?****

TheShylorchicds, 2025-11-24:
是的，我也准备用mid70尝试复现，目前有些疑惑，想问下大佬这个先进行mid70和外置imu标定和时间同步 步骤吗，然后在这个基础上运行后续的算法