hku-mars/FAST-LIVO2 #310: Mapping fails immediately on Jetson Orin Nano with OS0 LiDAR (works fine on x86)

TheOnlyLoki, 2025-06-16:
Hi @xuankuzcr,

First of all, thanks a lot for the great work on FAST-LIVO2 — really impressive project!

We’re running into an issue on ARM platforms:
On a Jetson Orin Nano (8 GB) with an Ouster OS0-128, the mapping completely fails. As soon as the system moves, the trajectory just drifts off into nowhere. Using the exact same code and parameters on an x86 machine, everything works perfectly.

Setup:
Platform: Jetson Orin Nano (Ubuntu 22.04, ROS2 Humble, custom fork)

Comparison: x86_64 Desktop (Ubuntu 22.04, ROS2 Humble)

Sensor: Ouster OS0-128 (including IMU)

Code: Latest FAST-LIVO2 main branch, no modifications

Parameters: identical on both systems (extrinsics, LIO, VIO, etc.)

Observations:
- All ROS2 topics (IMU, points, images) are published normally
- No noticeable delays or packet loss
- CPU and RAM usage on Jetson are within normal range
- IMU data comes directly from the OS0 sensor
- Pose drifts heavily immediately after first movement, as if IMU data is totally off — but only on Jetson

Additional info:
The problem also occurs when img_en: 0 (image input disabled)

Suspected causes:
It might be an ARM-specific issue (floating point, timestamp handling, Eigen library, etc.)
Or some synchronization / data processing on the Jetson with OS0 data is going wrong. On x86, everything is stable.

We’re using ROS2 Humble on Jetson Orin Nano. The mapping breaks down immediately upon first movement, while the PC with the same setup runs flawlessly.

Are there any known issues or special configurations required for ARM64 + Ouster with FAST-LIVO2?

Thanks and best regards.

xuankuzcr, 2025-06-18:
Hi,
Thanks for the detailed info! 
Just to confirm — FAST-LIVO2 works normally on ARM platforms, including Jetson. We've tested it successfully on several ARM64 devices. From your description, this looks like a data sync or message drop issue, not a platform bug.

TheOnlyLoki, 2025-06-22:
Hi,
My professor suggested that it could be that all of the Data we are getting from our camera and OS0 lidar + the data that the SLAM publishes are too much to handle for the system-bus of the jetson, so the jetson maybe drops messages. Do you think this might be the issue?

WenxuanLi-whu, 2025-09-10:
用什么板子效果运行流畅，推荐一下啊