hku-mars/FAST-LIVO2 #325: 请问livo2中的视觉部分仅是为了点云上色吗？

guxinglei911, 2025-07-04:
请问livo2中的视觉部分仅是为了点云上吗？还是说在雷达退化场景视觉提供匹配输出使得系统更加鲁棒. 我试了下：
纯激光可以跑，但是容易旋转偏差。 
激光+imu 可以稳定跑 
激光+视觉+imu 可以稳定跑，并且输出带颜色的点云很酷炫，但是整体点云效果和激光+imu没有对比出来。有大佬可以解答下疑问吗？

20251181, 2025-07-04:
请问您使用的传感器型号分别是什么呀？

guxinglei911, 2025-07-04:
主要是跑的提供的数据集对比，效果很好，但是没有对比出是否加入相机的点云区别  
我自己跑是了下velodyne16 + d435 ，效果一般，主要应该是外参标定和相机的有些问题

supersupercoconut, 2025-07-08:
> 主要是跑的提供的数据集对比，效果很好，但是没有对比出是否加入相机的点云区别 我自己跑是了下velodyne16 + d435 ，效果一般，主要应该是外参标定和相机的有些问题

我自己用的435i跟avia，效果也是一般，不知道是不是我自己外参弄的有问题

supersupercoconut, 2025-07-11:
> > > 主要是跑的提供的数据集对比，效果很好，但是没有对比出是否加入相机的点云区别 我自己跑是了下velodyne16 + d435 ，效果一般，主要应该是外参标定和相机的有些问题
> > 
> > 
> > 我自己用的435i跟avia，效果也是一般，不知道是不是我自己外参弄的有问题
> 
> 我用的Orbbec Gemini3335L和mid360效果非常好。可以从外参标定方面检查，严格使用优质的标定板，我用的是5mm铝合金的，识别码别用普通的纸，遇到胶水容易变形。再就是435i不是全局快门可能也是原因之一。

我手边没有那么好的标定版，所以我的操作方法就是用的无目标的标定方法（https://github.com/koide3/direct_visual_lidar_calibration），D435i确实不是全局快门，而且我是安装在车上，D435i与avia之间的外参可能受到车震动的影响不够准确，D435i还没有弄成全局快门，我是手动做的软件同步

BetterYimi, 2025-07-13:
> > > > > 主要是跑的提供的数据集对比，效果很好，但是没有对比出是否加入相机的点云区别 我自己跑是了下velodyne16 + d435 ，效果一般，主要应该是外参标定和相机的有些问题
> > > > 
> > > > 
> > > > 我自己用的435i跟avia，效果也是一般，不知道是不是我自己外参弄的有问题
> > > 
> > > 
> > > 我用的Orbbec Gemini3335L和mid360效果非常好。可以从外参标定方面检查，严格使用优质的标定板，我用的是5mm铝合金的，识别码别用普通的纸，遇到胶水容易变形。再就是435i不是全局快门可能也是原因之一。
> > 
> > 
> > 我手边没有那么好的标定版，所以我的操作方法就是用的无目标的[标定方法](https://github.com/koide3/direct_visual_lidar_calibration) ，D435i确实不是全局快门，而且我是安装在车上，D435i与avia之间的外参可能受到车震动的影响不够准确，D435i还没有弄成全局快门，我是手动做的软件同步
> > 一开始我也没有条件做标定，koide3我实在是不会...就砸钱了，呵呵，花了900买了一块铝板。200买了个承重好的支架，价格有些贵，但是帮助确实大。当初还怀疑相机的标定跟出厂设置不符合，买了600的棋盘格。总之就是换了联合标定板以后就标定的非常好了。点云效果立马就变得非常出色。而且我没有做任何同步，就是各自输出10hz。下一步打算加入同步看效果能不能更好。

@jiabotao 你好 请问你900块那个铝板标定板是怎么样的呀？方便发个淘宝链接吗

TokyoWarfare, 2025-11-08:
I wonder if it would be possible to use Livo 2 for the coloring steps and beyond, that is
-colorize point cloud.
-obtain camera poses (you could later use theese in SFM like Metashapet or similar, or to generate 3DGS in other softwares)
-obtain 3DGS

And do so by importing a bigger geopositioned data that has been pre procesed with something like
-https://github.com/PJLab-ADG/Livox-Mapping
-https://github.com/JokerJohn/MS_Mapping_2.0
-https://github.com/Yixin-F/better_fastlio2

The idea is to have the features of a mapping pipeline with GPS corrections, multisession, loop closure etc...,  so a bast area can be captured.
And the nice coloring, meshing  and other features of LIVO2, not relying ont he SLAM for positioning the data.

Wonder if anyone has tried similar of has an idea on how to achieve  this. 

Cheers!