hku-mars/FAST-LIVO2 #350: 静止点云上色清晰，运动模糊

zynzzz, 2025-08-11:
郑博你好，非常感谢您出色的工作！
我在复现过程中遇到一些问题想向您请教一下，我完成时间同步和标定后，Fastlivo2静止时上色效果非常好，但是一旦开始移动就会逐渐越来越模糊，我们怀疑是标定的问题，但尝试标定了很多次也没有办法解决，我们相机内参用的是ros-noetic-camera-calibration，外参用的是Fast-calib，以下是我的外参标定结果以及移动静止效果：

[image]

[image]

[image]

[image]

iorileslie, 2025-08-26:
是不是时间同步的问题，用郑博的LIV_handhold要设置img_time_offset: 0.1

gengyd2021, 2025-10-30:
你好，请问解决这个问题了吗？我目前也是静止或者缓慢移动都很清晰，进行移动就会比较模糊

czp0204, 2026-01-31:
是时间同步的问题，调整Image imu和lidar之间每一帧消息的间隙，打印出来看看时间偏差