hku-mars/FAST-LIVO2 #356: ROS2 humble下的fast-livo2 + mid360 + hiki camera的同步问题

XeonHis, 2025-08-26:
非常感谢Fast-livo2的工作，我在ros2 humble下进行复现fast-livo2 + mid360 + hiki camera的时候遇到了一些问题，已经给mid360和相机做了硬同步，在实时跑fast-livo2的时候会出现 imu and lidar not synced的情况，且delta time会持续增加。查看了一下rqt_bag的数据
这个是现有系统搭建的rqt_bag数据

[image]

这个是官方retail street的rqt_bag数据

[image]

发现/livox/imu的分布密度有显著不一样，请问是imu这一块的数据问题么？导致imu 和lidar数据同步出现异常
mid360的驱动使用的是 https://github.com/xuankuzcr/LIV_handhold/tree/main/livox_ros_driver2

XeonHis, 2025-08-27:
已解决

closed 2025-08-27 by XeonHis

Epiphany-f, 2025-11-25:
@XeonHis 请问一下，你是怎么解决的，我也出现了这个问题，而且我的相机端出现了卡在第一帧的情况