hku-mars/FAST-LIVO2 #359: 相機時間為2025年，但lidar卻是2020年

ZHG169, 2025-09-02:
您好，我想請問為何會有這樣的問題?目前我是在AGX Orin上跑FAST-LIVO2，採用Basler camera搭配Livox AVIA，GPRMC訊號也是透過stm32f103c8t6進行發送，但卻遇到兩種情況。

第一種是lidar和camera的時間戳很接近，只有小數點後不同，這裡有成功跑起FAST-LIVO2
[ INFO] [1749362369.458285339]: Get LiDAR, its header time: 1749362369.429712
[ INFO] [1749362369.794794829]: Get image, its header time: 1749362369.833225

第二種是lidar和camera的時間戳出現差異，因為第一次成功後，只是再更換driver就沒再跑起FAST-LIVO2，後面重新更換stm32f103c8t6就跑出這樣的結果
 [INFO] [1756799558.503710780]: Get image, its header time: 1756799558.581444
 [INFO] [1756799558.508095489]: Get LiDAR, its header time: 1590193022.799437

我有看到有一篇也是在討論同樣的問題，但與我使用的相機不同，想問這樣會是相機的問題嗎? 有嘗試過把舊的timeshare刪掉再重跑一次，也是同樣的結果，這樣該如何解決?

maxiangchi, 2026-04-09:
模拟时间

yccckid, 2026-08-25:
I doubt that this sync method is just to modify the time stamp so that lidar and camera can work in the same time?