hku-mars/FAST-LIVO2 #366: 运行不显示

zonglinWang, 2025-09-14:
在运行官方数据集时，rviz不显示点云，终端报错如下：[parameter_blackboard-1] [INFO] [1757837835.484605085] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '17' parameters already!
[fastlivo_mapping-3] [ERROR] [1757837835.670873186] [laserMapping]: Cannot find value for parameter: parameter_blackboard.model, will try again.
[fastlivo_mapping-3] [ERROR] [1757837835.670982919] [laserMapping]: Cannot find value for parameter: parameter_blackboard.model, will try again.
[fastlivo_mapping-3] [ERROR] [1757837835.670993261] [laserMapping]: Cannot find value for parameter: parameter_blackboard.model, will try again.
[fastlivo_mapping-3] [ERROR] [1757837835.671000167] [laserMapping]: Cannot find value for parameter: parameter_blackboard.model, will try again.
[fastlivo_mapping-3] [ERROR] [1757837835.671006403] [laserMapping]: Cannot find value for parameter: parameter_blackboard.model, will try again.
[fastlivo_mapping-3] [ERROR] [1757837835.671012815] [laserMapping]: Cannot find value for parameter: parameter_blackboard.model, will try again.
[fastlivo_mapping-3] terminate called after throwing an instance of 'std::runtime_error'
[fastlivo_mapping-3]   what():  Camera model not correctly specified.
[ERROR] [fastlivo_mapping-3]: process has died [pid 17295, exit code -6, cmd '/home/wzl/fast_ws/install/fast_livo/lib/fast_livo/fastlivo_mapping --ros-args -r __node:=laserMapping --params-file /home/wzl/fast_ws/install/fast_livo/share/fast_livo/config/avia.yaml'].
[rviz2-4] [INFO] [1757837835.759641572] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1757837835.759925300] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1757837835.772097436] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1757837835.880048065] [rviz2]: Stereo is NOT SUPPORTED
请问有友友碰到这种情况，怎么解决的？

VVivyy, 2025-10-22:
我的能运行了，我写了一篇CSDN，你看一下【Ubuntu22.04及ROS2复现Fast_livo2，包含ROS2的安装、cmake的版本选择等等 - CSDN App】https://blog.csdn.net/LPKPLPK/article/details/153679798?sharetype=blog&amp;shareId=153679798&amp;sharerefer=APP&amp;sharesource=lpkplpk&amp;sharefrom=link

 
同问，我也用ros2跑的，改了几天都没用

HollowCrown-PH, 2025-10-22:
> 我的能运行了，我写了一篇CSDN，你看一下【Ubuntu22
非常感谢，已经成功了